jneilliii/OctoPrint-OctoDashCompanion #4: Webcam Command

fettesb, 2021-05-05:
Webcam did not show up with this Command
[!WEB]http://localhost/plugin/octodashcompanion/static/html/webcam.html

only when I set this Command
[!WEB]http://localhost/webcam/?action=stream

So my stream is working but it does not look good

[image]

jneilliii, 2021-05-05:
I've seen mixed results with the webcam iframe on two installs of mine. One works fine and the other one can't load pages at all in the iframe ie `[!WEB]https://www.google.com`. I haven't figured out why one can and the other can't yet. It just gets stuck on webpage loading...

Try my current dev version though to see. Install by copying/pasting the URL below into Plugin Manager > Get More > ...from URL and clicking Install. 
```
https://github.com/jneilliii/OctoPrint-OctoDashCompanion/archive/refs/heads/feature/dynamicwebcam.zip
```
Once installed change your Custom Action button to `[!WEB]http://localhost/plugin/octodashcompanion/webcam`. It should dynamically load the webcam stream from OctoPrint settings with all the flip/rotation options set.

jneilliii, 2021-05-05:
I just figured out the issue @fettesb. For the webcam page to be able to load inside OctoDash you have to enable iframe support in OctoPrint's settings. Under the `Server` section in settings, check the box to allow embedding, save, and then restart OctoPrint.

[image]

fettesb, 2021-05-05:
i will try that as soon as my print is done. Thank you

jneilliii, 2021-05-05:
I actually just updated the branch with a fix that should allow you to access the webcam without enabling that option. Use the same url to install, and change the custom action as mentioned before.

fettesb, 2021-05-05:
i am now on 0.0.3dev2 and set [!WEB]http://localhost/plugin/octodashcompanion/static/html/webcam.html
for my Webcam Command. After a reboot I got only "Webpage loading..."

jneilliii, 2021-05-05:
Awesome, thanks for confirming. I'll ne releasing stable version 0.0.3 once I get documentation updated.

closed 2021-05-06 by jneilliii