jneilliii/OctoPrint-OctoDashCompanion #6: Add GPIO?

DocZagreus, 2021-05-06:
Hi!
I love this plug-in!
Do you think could be possible to create some kind of action to toggle on/off a light connected to Octoprint via GPIO?
Right now I'm using octolight to control the light and would be great to be able to create a light button on octodash :)

jneilliii, 2021-05-06:
That is already possible with other plugins and gcode commands. You would just add those to the Custom Action. I have a Tasmota device with ws2812 on it that I control this way using my Tasmota plugin, but there are other plugins that can control just GPIO, ie enclosure, PSUControl, etc.

DocZagreus, 2021-05-06:
I'm sorry but I'm pretty new to this "world".
How can I add a custom button using octodash companion that allows me to switch on/ff my light?

jneilliii, 2021-05-06:
Looking at that OctoLight plugin, it looks like it doesn't support any form of gcode command for control. You may have better results with PSUControl to control that relay. It does support gcode commands to turn on and off the light. Then you would just add whatever gcode command that it uses in custom actions (I think the defaults are M80/M81, but are configurable), similar to this, replacing the gcode command with what that plugin expects.

[image]

jneilliii, 2021-05-06:
You would have to have one button for on and one button for off.

DocZagreus, 2021-05-06:
yep I was able to do it =)
Using LEDStripControl and using M150 W to turn on and M15o to turn off
thanks =)

DocZagreus, 2021-05-06:
Just one last question...
Is there a way to resize the icons so that I can have a 2 by 4 buttons layout that fits the monitor interface without scrolling?

jneilliii, 2021-05-06:
I'm not as familiar with the OctoDash internals to do this. It might be possible with a custom theme, but can't say I've seen one that messes with those buttons yet.

closed 2021-05-08 by jneilliii