jneilliii/OctoPrint-OctoDashCompanion #7: [Feature Request] Remote OctoDash

hawkeyexp, 2021-05-07:
Wow not noticed that you added this plugin - really great step forward!

Would be nice if you can find a solution to manage remote installs of octodash which would be great cause for example in my setup a rpi (gen 4) is fixed installed in my case and a second (possible outsorted older gen 3 ones) is running octodash!

hawkeyexp, 2021-05-07:
hi, yes it's a clean raspbian install and on the fixed rpi4 to control the printer octopi is installed :-)
I keep it splittet caused by construction reasons of my case with some neopixels servocontrols etc. and the rpi display is mounted outside of the case to get easy access :-)

My idea: i think you are wrinting to the filesystem where octodash writes it's config file correct?
An ugly solution would be configuring a nfs mount etc. to allow the plugin to reach the config file which is on the other device

jneilliii, 2021-05-08:
Do you have an octodash running on the pi4 as well? Just trying to get a total picture, because like you said, you could symlink the path on the pi4 over to the other pi's config path and that would work. Or also rsync or cron job script copy the file over to the other pi. I would be hesitant to come up with a script, but would be happy to implement the ability to create your own script to run after the config is saved. Then you could have the plugin send the file via scp file transfer to the other pi. I just need to look into how to implement the script running process, and figure out the best way to integrate into settings.

hawkeyexp, 2021-05-08:
Hi, my systems are fully splitted - the pi3 is only running octodash and the pi4 is running octoprint.
I think copy fils is not a really nice solution.
How about talking to Timon to integrate an api etc to transfer the new settings to the config? It also would be a benefit cause octoash can check the nw values to ensure the config files keeps valid.

jneilliii, 2021-05-08:
If an api were available on the octodash side I'd be happy to implement integration, but that's up to @UnchartedBull if he wants to implement a server side api for config access/modification.

UnchartedBull, 2021-05-08:
Probably a nice idea, especially for config validation. There are also some plans to introduce an API to query OctoDash logs to make debugging a lot easier. So it would make sense to introduce the new API together with those changes. The logger should be available for v3.1.0.

jneilliii, 2021-05-08:
It would really clean up the hacked approach I took with the plugin for loading and saving the config as well. I'll keep an eye out for those changes and adjust accordingly as those become available.

hawkeyexp, 2021-05-14:
Ok thx - for now i added passwordless ssh auth between both systems and modified your system command for start stop etc. to trigger a local shell script which copies the config and css via scp and does the start stop command also.
This way it's good useable for now.

stefan-sherwood, 2024-12-16:
I have the identical problem. My OctoPrint is in the room with the printer and my OctoDash is in my office where I can watch the status of a print. I don't think this is an uncommon setup.