jneilliii/OctoPrint-OctoDashCompanion #14: [Feature Request] Add slect cam to action command

hawkeyexp, 2021-05-24:
Please add an option to make it possible to select which cam should be shown by the action. If nothing is set use the current solution with select via octodash on preview.

jneilliii, 2021-05-25:
added with the above commit. you'll need to manually reinstall from the rc branch if you want to test it out. The url pattern to use is as follows, adjust for your OP URL of course. The number is based on where it is in the list from multicam view settings.

```
http://localhost:5000/plugin/octodashcompanion/webcam?webcam=2
```

hawkeyexp, 2021-05-25:
you are amazing fast :-) i will try it out after current print is finished :-)

hawkeyexp, 2021-05-25:
i have tested it and the only thing i get is an internal server error while trying to reach the cam with the select - without ?webcam=x it works as before.

jneilliii, 2021-05-25:
can you check your octoprint.log for the error that it's generating please?

hawkeyexp, 2021-05-25:
sure i will add it after current print has finished as usual :-) ... seems i ned a second printer ... Voron i'm coming...

hawkeyexp, 2021-05-25:
Here is the log:
```

2021-05-25 21:03:55,987 - octoprint.server.util.sockjs - INFO - User root logged in on the socket from client ::ffff:192.168.188.211
2021-05-25 21:04:02,926 - octoprint - ERROR - Exception on /plugin/octodashcompanion/webcam [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octodashcompanion/__init__.py", line 102, in webcam_route
    if webcam["flipH"]:
KeyError: 'flipH'
```

source lin was correctly added to config.json:

`http://192.168.188.210/plugin/octodashcompanion/webcam?webcam=1`

Also checked with webcam=0 (i'm using a raspicam and a usb cam)

jneilliii, 2021-05-25:
thanks, might have to change the logic on those checks maybe to `if "flipH" in webcam` can you share the multicam section of your config.yaml? It makes me wonder if your settings might be corrupted or something. definitely need to add some more logging, during the process I think, because it might not be getting a hold on the specific entry for the webcam.

hawkeyexp, 2021-05-26:
here we go:

```
    _config_version: 1
    multicam_profiles:
    - URL: http://192.168.188.210/webcam/?action=stream
      isButtonEnabled: false
      name: Default
      snapshot_url: http://192.168.188.210/webcam/?action=snapshot
    - URL: http://192.168.188.210/webcam/?action=stream
      isButtonEnabled: true
      name: RaspiCam
      snapshot_url: http://192.168.188.210/webcam/?action=snapshot
    - URL: http://192.168.188.210/webcam2/?action=stream
      isButtonEnabled: true
      name: Nozzle
      snapshot_url: http://192.168.188.210/webcam2/?action=snapshot

```

jneilliii, 2021-05-26:
interesting...that may be why it's erroring for sure. that's odd those flipH, flipV, and rotate90 fields aren't there for you and then are for me. Curious if you configure those options once and save, and then re-configure to the way you need them, do those fields get added to the settings? That would also resolve the error.

hawkeyexp, 2021-05-26:
I'm sorry to say but problem was on my side and didn't notice :-(
During the play arround with my gui lyouts etc. i installed a fork of multicam and didn't remember - this fork didn't write the full config :-(
Now reinstalled the correct one and it works without problems!

closed 2021-05-26 by hawkeyexp