jneilliii/OctoPrint-OctoDashCompanion #17: ambientSensorID

ssavickiy, 2021-06-23:
Hello. I have the enclosure plugin installed, the dht22 temperature and humidity sensor is configured, but I do not know what should be specified in the ambientSensorID for octodash to work correctly? How do I find out this id?

jneilliii, 2021-06-23:
I'm not really sure on how those settings work as I don't use the enclosure plugin, just added the settings as I saw them in the config.json.

jneilliii, 2021-06-23:
I just took a quick peak at the enclosure plugin and it seems the index is an arbitrary number added by the plugin when you add the sensors. So in theory counting from top to bottom starting at 1.

ab-tools, 2022-02-28:
You can just access the API URL in a browser:
http://octopi.local/plugin/enclosure/inputs

`index_id` will be the ID you are looking for.

closed 2022-03-20 by jneilliii