jneilliii/OctoPrint-OctoDashCompanion #21: Predefined Action

iwTao, 2021-08-21:
Hi,jneilliii,if I want add or adjust predefined action, such as define a new action [!reboot klipper].what should I do, or which file should I modify?

jneilliii, 2021-08-21:
seems like that would be something that could be done as a macro within klipper and then you would just use that command as gcode in your button?

iwTao, 2021-08-21:
> seems like that would be something that could be done as a macro within klipper and then you would just use that command as gcode in your button?

no, i just want click the button, then klipper will restart, just like i write “systemctl restart klipper”in my Raspberry PI’s terminal .so i want know how add a predefined action.

jneilliii, 2021-08-21:
gotcha. you might find the [GCODE System Commands](https://plugins.octoprint.org/plugins/gcodesystemcommands/) plugin handy for this then. You define a custom gcode command like OCTO1 as `service restart klipper` and then configure your custom action as `OCTO1`.

iwTao, 2021-08-22:
Tkanks for your help, i was set OCTO1 command, but i  receive `Return(GCodeSystemCommands): error`  when i send COTO1 in my octoprint terminal.

jneilliii, 2021-08-22:
Try seeing it to `sudo service restart klipper`

iwTao, 2021-08-24:
I resolved this problem,need add "bash" in front of .sh file directory.

closed 2021-08-24 by jneilliii