jneilliii/OctoPrint-OctoDashCompanion #25: FileNotFoundError No such file or directory: '/home/pi/.config/octodash/config.json'

joaquinabian, 2022-01-17:
**This is:**
OctoPrint version : 1.7.2
OctoPi version    : 0.18.0 (Linux octopi 5.10.63-v7l+)
Dashboard 1.19.6
OctodashCompanion 0.6.6
on a Raspberry Pi 4

**Problem**
I had a problem with connectivity and trying to fix it I discovered that the octoprint.log file had several **FileNotFoundErrors** from OctoDashCompanion.
It searches the Octodash file config.json at  '/home/pi/.config/octodash/config.json' but  '/home/pi/.config' doesn't exists on the raspberry.

In the Octoprint OctoDash Companion Settings page, there is a note saying: 

> The below settings are assuming OctoDash's config.json file is located at --- and running on the same machine as this OctoPrint instance.

but the space where the path should be is empty.

I attach a section of the log with the Error and an image of the settings page.

[octoprint.log.section.txt](https://github.com/jneilliii/OctoPrint-OctoDashCompanion/files/7882014/octoprint.log.section.txt)
[image]

jneilliii, 2022-01-17:
Does the file exist where it supposed to or not? If OctoDash has never been installed and initialized with settings then this would be the reason why the plugin doesn't see the file.

joaquinabian, 2022-01-19:
Sorry, close.
I confused OctoDash with Octoprint Dashboard.

closed 2022-01-19 by joaquinabian