jneilliii/OctoPrint-OctoDashCompanion #27: Cannot Remove Sensor ID Again

ab-tools, 2022-02-28:
Steps to reproduce:
1. Enter `1` as Ambient Sensor ID on Plugins tab.
2. Save setting.
3. Open settings again and remove Ambient Sensor ID again.
4. Save settings.

The previously entered sensor ID is still shown and cannot be removed again.

jneilliii, 2022-02-28:
I'll have to check on this one. It's either not loading from the config.json correctly or not saving back correctly.

jneilliii, 2022-03-01:
Yeah, I believe this is because of how I'm merging the original file with changes coming from OctoPrint's settings interface. Since I'm using octoprint.util.dict_merge with the two dictionaries it doesn't see blank as an actual value and therefore does not remove the value and the value from config.json persists. Not sure how I'd get around this right now.