jneilliii/OctoPrint-OctoDashCompanion #34: When Octopi Is installed on Docker and OctoDash Companion is installed I get the following error in the container logs:

hiramknick, 2022-10-12:


jneilliii, 2022-10-12:
yeah, the plugin assumes you are running octodash on the same device that octoprint is running on and not in a docker container.

hiramknick, 2022-10-12:
In this case both Octoprint and Octodash are both running in the same container.  Effectively I have a Raspberry Pi with a Docker Container for the Octoprint Subsystem including all plugins.  I can attach the docker Compose file I am using (In transparency I have 2 Octoprint instances, one for each of 2 printers.  

I'm already working on adding volumes to handle OS dependencies from the host but attached is my current draft for my 2 printers (FLSUN QQSPro and FLSUN SuperRacer)

When trying to browse to the file in root within the docker (this was not migrated from bare metal to docker, its always been run in docker) the folder doesn't exist.  I'll try spinning up a third instance and installing the plugin to see what the error logs look like.

[Octoprint_Basic.txt](https://github.com/jneilliii/OctoPrint-OctoDashCompanion/files/9764444/Octoprint_Basic.txt)

hiramknick, 2022-10-12:
When I do a fresh docker image with nothing else installed I see the following: 

Here are the logs during install:
```
2022-10-12 11:53:19,740 - octoprint.plugins.pluginmanager - INFO - Installing plugin from https://github.com/jneilliii/OctoPrint-OctoDashCompanion/archive/master.zip
 2022-10-12 11:53:20,253 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin (2022.6.13) did not pass check, not loading.
 2022-10-12 11:53:29,566 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin (2022.6.13) did not pass check, not loading.
 2022-10-12 11:53:29,598 - octoprint.plugins.pluginmanager - WARNING - The plugin was installed successfully, but couldn't be found afterwards to initialize properly during runtime. Please restart OctoPrint.
 2022-10-12 11:54:58,711 - octoprint.server.api.system - INFO - Performing command for core:restart: s6-svc -r /var/run/s6/services/octoprint
 2022-10-12 11:54:58,751 - octoprint.server - INFO - Shutting down...
 2022-10-12 11:54:59,414 - octoprint.server - INFO - Calling on_shutdown on plugins
 2022-10-12 11:54:59,417 - octoprint.events - INFO - Processing shutdown event, this will be our last event
 2022-10-12 11:54:59,427 - octoprint.events - INFO - Event loop shut down
 2022-10-12 11:54:59,467 - octoprint.server - INFO - Goodbye!
```
```
 2022-10-12 11:55:08,042 - octoprint.plugins.octodashcompanion - WARNING - The Blueprint of this plugin is relying on the default implementation of is_blueprint_csrf_protected (newly added in OctoPrint 1.8.3), which in a future version will be switched from False to True for security reasons. Plugin authors should ensure they explicitly declare the CSRF protection status in their BlueprintPlugin mixin implementation. Recommendation is to enable CSRF protection and exempt views that must not use it with the octoprint.plugin.BlueprintPlugin.csrf_exempt decorator.
 2022-10-12 11:55:08,046 - octoprint.server - WARNING - CSRF Protection for Blueprint of plugin octodashcompanion is DISABLED
```

```
 2022-10-12 11:55:13,050 - octoprint.plugins.octodashcompanion - INFO - attempting open of /root/.config/octodash/config.json.
 2022-10-12 11:55:13,073 - octoprint.server.api.settings - ERROR - Could not load settings for plugin OctoDash Companion (0.0.7)
 Traceback (most recent call last):
   File "/usr/local/lib/python3.8/site-packages/octoprint/server/api/settings.py", line 372, in _get_plugin_settings
     result = plugin.on_settings_load()
   File "/usr/local/lib/python3.8/site-packages/octoprint/util/__init__.py", line 1688, in wrapper
     return f(*args, **kwargs)
  
 File "/octoprint/plugins/lib/python3.8/site-packages/octoprint_octodashcompanion/__init__.py", line 45, in on_settings_load
     with open(self.config_file) as file:
 FileNotFoundError: [Errno 2] No such file or directory: '/root/.config/octodash/config.json'
```

It looks like the default saving location for octoprint (to make it persist across boots) in Docker is (local path)/home/octoprint/.octoprint that way the settings are saved to persistent disk.

jneilliii, 2022-10-12:
Yeah, the plugin is using the call `config_directory=normalize(_default_configdir("octodash"))` to determine it's location, which I guess is not picked up properly in docker or not set the same.

hiramknick, 2022-10-13:
Issue raised with the Docker container:  https://github.com/OctoPrint/octoprint-docker/issues/240

jneilliii, 2022-10-13:
The is actually an internal function that expands the path using `os.path.expanduser(os.path.join("~", ".config", applicationName.lower()))` in linux environments, and I believe since the docker container is running as root that path is getting interpreted properly. I can program around this via manual updating of the settings, I just need to verify that the code is referencing the settings instead of this internal function call. You can probably the close the issue on the other repo.

jneilliii, 2022-10-13:
How does the octodash component get installed? I don't see any reference in your example compose file for that, did you manually install it after booting up the container?

hiramknick, 2022-10-13:
I added it from the plugins module in the settings manually.

jneilliii, 2022-10-13:
the plugin alone isn't enough to use for touch screen access. you also have to install [OctoDash](https://github.com/UnchartedBull/OctoDash). Are you sure you aren't looking for the [dashboard ](https://plugins.octoprint.org/plugins/dashboard/) plugin?

hiramknick, 2022-10-13:
> I have both of them in the plug-ins manually added.

How I found this is that with your plugin installed the streaming plugin
keeps crashing and taking down the octopi interface requiring a refresh in
the docker. I found a couple of other causes but have been able to fix them
by linking libraries to the docker as shared volumes.

closed 2022-11-05 by github-actions[bot]