jneilliii/OctoPrint-OctoDashCompanion #42: Using Octodash with either GPIO or Tasmota

Ecks32, 2025-07-06:
I'm trying to control a light in my shop with a relay, I need a control both in the shop and on my PC in my house.  

I have tried the GPIO plugin but I can't find a way to get a control in Octodash, the buttons only show in the web interface.  Is there a way to add a Custom Actions or any type of button to toggle a GPIO pin, either with or without the GPIO plugin?

I've also tried the Tasmota plugin but I'm not understanding how it works.  As far as I can tell it's designed to only respond to printer events. Is that correct?  Or is there a way to toggle a Tasmota relay with the Custom Action buttons?  

I have been able to use the Custom Action Buttons to send a Tasmota HTTP command but I can't get Toggle to work so I have to have a button for both On and Off.  When I have it send Toggle it will only send once, then the button doesn't seem to work again until I go to the home screen and back.  If I have it set to go to the home screen as part of the sequence then the Toggle command doesn't work at all.  It will simply go to the home page without triggering the relay. 

I have a feeling the answers are likely no and no but if one of these is possible please let me know how it's done.  Since Tasmota isn't the most reliable I would rather use the GPIO pins, but whatever works.

Thanks

jneilliii, 2025-07-06:
> I have tried the GPIO plugin but I can't find a way to get a control in Octodash, the buttons only show in the web interface. Is there a way to add a Custom Actions or any type of button to toggle a GPIO pin, either with or without the GPIO plugin?

does that plugin offer a way to trigger using gcode commands?

> I've also tried the Tasmota plugin but I'm not understanding how it works. As far as I can tell it's designed to only respond to printer events. Is that correct? Or is there a way to toggle a Tasmota relay with the Custom Action buttons?

My Tasmota plugin can be configured to be controlled via gcode commands, and you would add those gcode commands to OctoDash custom actions to get the buttons in it's interface. This is actually what I do. Only issue here is you would need two buttons like you mentioned, one for off and one for on. 

You might be able to get around this using the PSU Control plugin directly with GPIO. For toggle functionality you would have to also use GCODE System commands configured with an API call to the togglePSU endpoint using curl. https://github.com/kantlivelong/OctoPrint-PSUControl/wiki/API#curl-examples

Ecks32, 2025-07-07:
> does that plugin offer a way to trigger using gcode commands?

No, looks like it only provides buttons in the web interface.

> My Tasmota plugin can be configured to be controlled via gcode commands,

That's what I've been missing.  It hadn't occurred to me to use gcode for anything but controlling the printer, and is why I was under the impression the plugin only responds to printer events.  I guess I would make up gcode that isn't assigned to anything else? or would they have to be existing commands but something my printer doesn't recognize?  I'll look up a list of gcode commands and figure it out.   

This opens up a lot of possibilities since so many plugins seem to work with gcode.  I looked at the PSU plugin but also couldn't find a way to be able to control it from both places, but now I'll need to look at it again.  I may go ahead and stick with the Tasmota http commands.  It occurred to me that if I use a Tasmota plug then I'd also have a physical button to control the light and could still control it from my PC if the Raspberry Pi isn't powered on.  

While looking through all the available plugins I noticed you write a lot of them.  I appreciate everything you do to make 3d printing easier for everyone.  Supporting your work has been well worthwhile.  Thank you.